systemsomicslab/MsdialWorkbench #63: Debugging error even after following the installation guide in Visual Studio

Soumya624, 2023-03-24:
[image]

kozo2, 2023-03-26:
@Soumya624 Reloading the project may be an effective solution to that.
You can use the keyboard shortcut Ctrl + Shift + F5 to rebuild and reload all projects in your solution.

You can also reload the project with
1. Open Solution Explorer
2. Right-click on the projects in your image's "Project" column.
3. Select "Reload Project" from the context menu.

I hope that helps! Let me know if the above does not solve the problem.

Soumya624, 2023-03-26:
Hi @kozo2 ,

Thanks for your response. It's solving the problem.

Also, i was interested in **A commandline application of MS-DIAL** project for this year's GSoC. So, is it still open for application?

Thanks

kozo2, 2023-03-26:
Yes, it's still open.

1. Write your proposal (there is a template in https://github.com/MZmine-MS-DIAL-GSoC/MZmine-MS-DIAL-GSoC.github.io/discussions/13 ).
2. Submit your proposal in https://summerofcode.withgoogle.com/ .

closed 2023-03-26 by kozo2