systemsomicslab/MsdialWorkbench #89: Implementation of Parameter export function

kozo2, 2023-05-22:
Implement the functionality of the red circle in the attached image in MS-DIAL version 5.

[image]

kozo2, 2023-09-12:
Hi @FunkiiL 
We have uploaded a provisional version of MS-DIAL5 with the enabled "Parameter export button".
https://github.com/systemsomicslab/MsdialWorkbench/releases/tag/issue206

We would appreciate it if you could provide us with feedback (if you have any thoughts).

yangyi-pku, 2023-10-08:
Hi @kozo2,
We have exported the parameters in .txt format, but how can I load these parameters in this format when I start a new project? It seems that I can only load parameters in the .mdparameter format.  
thank you!