systemsomicslab/MsdialWorkbench #119: MS-DIAL crushes for unknown reason - Windows

olgafccps, 2023-06-20:
Hi!
It seems we need some help with MS-DIAL software, Windows version. 
The attached file is a screenshot from version 5.1.230517 and when we press the "Next" button on that screen the program thinks for some seconds then closes without any error messages. We also have tried version 4.9.221218 but the effect was similar. The computer where it all happens has 32 Gb RAM, last generation of i7 CPU, lots of free disk space and Win10 Pro.  Turning off the antivirus software does not help. 
My colleague runs the software successfully on a similar computer. Can you suggest any ways to solve the problem?

Thank you in advanced!
[image]

kozo2, 2023-07-26:
Hi @olgafccps ,
We recently uploaded a new version `5.1.230719` to http://prime.psc.riken.jp/compms/msdial/main.html . Can you tell me if your problem is reproduced with `5.1.230719`?

kozo2, 2023-08-24:
Hi @olgafccps Since I haven't received a response for a while, I will temporarily close this issue. If the problem persists, please feel free to reopen it.

closed 2023-08-24 by kozo2

arpita-007, 2024-02-03:
Hi,
I am facing the similar issue. I uploaded 80 files and when I start the alignment, after 62-69/80 it gets closed without showing any error. I am using the 5.1.230912 version on 64 GB system with 12 cores.
Thank you in advance!

arpita-007, 2024-02-03:
> Hi @olgafccps Since I haven't received a response for a while, I will temporarily close this issue. If the problem persists, please feel free to reopen it.

Hi @kozo2 ,
I am facing the similar issue. I uploaded 80 files and when I start the alignment, after 62-69/80 it gets closed without showing any error. I am using the 5.1.230912 version on 64 GB system with 12 cores.
Thank you in advance!