systemsomicslab/MsdialWorkbench #126: MS-Finder works only when Windows display language is set to english.

osowiecki, 2023-06-29:
MS-finder fails to start when system display language in Windows 10 is set to polish. Works fine when it's set to english. Ms-dial doesn't have this problem.

kozo2, 2023-06-29:
@osowiecki Thanks for your new issue.
Can I ask you the MS-finder and Ms-dial version?

osowiecki, 2023-06-29:
MSDIAL ver.4.9.221218 Windowsx64
MS-FINDER ver. 3.52 Windows

Checked on 3 other machines with the same result. Display language != English is a crash.

kozo2, 2023-07-04:
@osowiecki 
You are right, it does seem to crash in non-English.
But our development team does not have the time to check non-English language environment.
I don't think we can raise the priority of this matter at this time.

I think the issue will remain open.
I would like to ask you to switch to English.

closed 2024-01-29 by kozo2