systemsomicslab/MsdialWorkbench #128: Reference for Demo Data

Soumya624, 2023-07-01:
Hi @kozo2,

Is there any dummy .wiff available for testing? I am using the current version of MS-Dial. But, couldn't find any demo folder for mass spectrometry data. Any link/reference to the achieve would be great.

Thanks!

kozo2, 2023-07-11:
Hi @Soumya624 
Sorry for the late reply.

>  Is there any dummy .wiff available for testing?

We don't have a dummy .wiff, but we do have demo .wiff files for testing.
http://prime.psc.riken.jp/compms/msdial/main.html#:~:text=CCS%20database).-,Demonstration%20files,-Lipid%20profiling%20of

> But, couldn't find any demo folder for mass spectrometry data.

Sorry, but I don't think we will be adding demo files to reduce the data size of our releases.
We will be doing future releases in GitHub releases.
GitHub releases has a 2GB limit on file size.

> Any link/reference of the archive in the README.md would be highly helpful!

Thanks for the comment.
I will do it.

Soumya624, 2023-07-15:
Hi @kozo2 ,

Thanks a lot for the link. It's working fine from my end!

closed 2023-07-15 by Soumya624