systemsomicslab/MsdialWorkbench #166: Class ID and Sample Type Assignment in Console App

AbrahamProject, 2023-08-07:
Hi,

I am currently trying to use the MsdialConsoleApp.exe from version 4.9.221218 .  I was able to set all the needed parameters in a txt file, however I am not sure how to assign the sample type (Blank/QC/Samples) and the Class id to different the different files in order to analyse biological groups using the method file for the command line (It would be of relevance for various settings in the alignment step). 
If there is such possibility to assign it through the method file I would appreciate any help 

Thanks a lot

P.S. Thanks for all the effort in making a user friendly Application for Metabolomics/Lipidomics

Sincerely,
Abraham

YukiMatsuzawa, 2023-08-10:
Hi @AbrahamProject,

Sorry for the late response.

There is a feature to specify ClassID and SampleType. However, please be aware that there is a bug in the latest version 4.9 that prevents it from working. Please consider trying this version of MSDIAL that I've attached here. You can make use of the attached CSV file by specifying it with the -i option. This will allow you to set these parameters according to the format provided in the CSV.

Thank you.

[Msdial](https://drive.google.com/file/d/18NVlDMQF3VXqUcTE40hZ_cPFUO-MiFY_/view?usp=drive_link)
[classid_sample.csv](https://github.com/systemsomicslab/MsdialWorkbench/files/12313736/classid_sample.csv)

AbrahamProject, 2023-08-15:
Hi,

Thanks so much will give it a try

closed 2024-01-29 by kozo2