systemsomicslab/MsdialWorkbench #168: Revert old column-wide export format in MS-FINDER.

kozo2, 2023-08-07:
The export format of MS-FINDER has become row-wide in version 3.6.0.

However, due to user requests, we will add an option to export the old column-wide format.

gegoe76, 2023-08-07:
Hi Kozo, ok I have subscribed to the project.

Cheers, Geert

kozo2, 2023-08-28:
Hi @gegoe76 ,
I have attached the MS-FINDER with the feature you wanted restored in https://github.com/systemsomicslab/MsdialWorkbench/releases/tag/test_widerlonger_selectable_export. Could you please confirm if there are no issues?

kozo2, 2023-09-07:
Hi @gegoe76 ,
Have you tried https://github.com/systemsomicslab/MsdialWorkbench/releases/tag/MSFINDER-v3.71 ?
Does it have the features you want?

gegoe76, 2024-01-03:
Hi,

I'm sorry but I only saw this message now....
Anyway, i tried it straight away, the export looks good now when choosing wider table.

BUT: 

1. i can not upload my in-house libraries anymore, the program shuts down when trying to do this, can you fix this please?
2. it still shows as version 3.57 and not as version 3.71, which is confusing because i have many different versions now.

How can i upload my inhouse libraries? It's not possible anymore.

thanks, 
cheers, geert

gegoe76, 2024-01-03:
also, i don't know why, but it takes much longer now to load the online databases, 10 times longer than before...

why is this?

gegoe76, 2024-01-03:
i was able now to upload user defined database, i don't know what went wrong, but now it works

closed 2025-05-15 by Bujee415