systemsomicslab/MsdialWorkbench #172: MSDIAL crashes for no peak information during GC-MS alignment

QizhiSu, 2023-08-08:
Dear MS-DIAL developers,

I am currently using MS-DIAL version 4.90 for GC-MS data processing. 
The software crashed during the alignment step. We have noticed that during the peak detection step, some samples have not been run, and if we unticked "Together with alignment" and when we try to open those unrun samples, we get "There is no peak information in 'the sample'. So please select other files or check ion mode for it and re-analyzed it with the ion mode setting". We have tried to lower the "Minimum peak height" to even 10 amplitude, the problem remain. 
[image]
[image]

The weird thing is that when we select a smaller subset (including those unrun samples), it works perfectly. Besides, we can open those files by Chemstation with no problem. which means that the data files should be fine. 

Another thing is that when we try different times, different files have problems. Sometimes files A and B have problems, othertimes, files C and D have problems, which makes me quite confused. 

in our experience, similar thing happened sometimes but not always. 

We also tried Version 4.7, 4.8 and other computers, the problem remains.

Any clue?

Thanks a lot, in advance. 

Sukis

YukiMatsuzawa, 2023-08-09:
Hi @QizhiSu,

Currently, we're uncertain about the exact cause. If you could provide the file where the error is occurring, we might be able to conduct a thorough investigation into the issue.

Thank you.

QizhiSu, 2023-08-09:
Hi @YukiMatsuzawa ,

Thanks a lot for your prompt response. We then unticked identification after alignment, the issue has gone. So, it is fine now. 

All the best, 
Sukis