systemsomicslab/MsdialWorkbench #174: are raw data and results saved in a server?

vasiloa2, 2023-08-08:
Dear MS-DIAL developers, 

I would like to use MS DIAL for processing and analysis of a big project where data are confidential. My question is whether the raw data files I upload and the projects I create, are saved in a server or any other space other than locally to my PC? 
I run the .exe file every time that I process a dataset. Would it make any difference to install the app - would that make a difference where data are stored? Is there a dosumentation for my question?

My best regards
Katerina

kozo2, 2023-08-10:
Hi @vasiloa2 ,

> My question is whether the raw data files I upload and the projects I create, are saved in a server or any other space other than locally to my PC?

No data is saved anywhere other than locally.
All data is saved only under the MS-DIAL project file path.

[image]

> I run the .exe file every time that I process a dataset. Would it make any difference to install the app - would that make a difference where data are stored?

It makes no difference.

> Is there a dosumentation for my question?

We have not created a privacy policy document yet.
You can check the progress in https://github.com/systemsomicslab/MsdialWorkbench/issues/177 .

kozo2, 2023-08-10:
Hi @vasiloa2 ,

> My question is whether the raw data files I upload and the projects I create, are saved in a server or any other space other than locally to my PC?

No data is saved anywhere other than locally.

> I run the .exe file every time that I process a dataset. Would it make any difference to install the app - would that make a difference where data are stored?

It makes no difference.

> Is there a dosumentation for my question?

We have not created a privacy policy document yet.
You can check the progress in https://github.com/systemsomicslab/MsdialWorkbench/issues/177 .

vasiloa2, 2023-08-10:
Dear Kozo Nishida,

thank you very much for your prompt and helpful reply!

Best regards
Katerina

-- 
*Dr. Katerina Vasilopoulou*
Post-doc Analytics

Roche Diagnostics GmbH
HDHHBE7A
Nonnenwald 2
82377 Penzberg / Germany

***@***.*** ***@***.***>

*Roche Diagnostics GmbH*
Werk Penzberg; Nonnenwald 2; D‑82377 Penzberg; Telefon +49‑8856‑60‑0;
Telefax +49‑8856‑60‑3896
Sitz der Gesellschaft: Mannheim - Registergericht: AG Mannheim HRB 3962 -
Geschäftsführung: Claus Haberda; Andreas Schmitz -
Aufsichtsratsvorsitzender: Dr. Thomas Schinecker

*Confidentiality Note*
This message is intended only for the use of the named recipient(s) and may
contain confidential and/or privileged information. If you are not the
intended recipient, please contact the sender and delete the message. Any
unauthorized use of the information contained in this message is prohibited.