systemsomicslab/MsdialWorkbench #181: Big difference in version EAD lipid identification 5.1.230222 vs 5.1.230807

Cagakan31, 2023-08-16:
Hi All,
I would like to highlight an important issue I recently faced, especially for EAD identification.
I am processing exactly the same sample which were acquired with sciex zeno tof 7600 DDA with EAD fragmentation. 
In the version 5.1.230222 many PE lipids were identified however in version 5.1230807 they are all gone. (Slide 1) I can see some of the PE species only if I click suggested. I can share the files and open for discussion.
Thanks for your time.
Best.
Cagakan

[MSDIAL 5 problem.pptx](https://github.com/systemsomicslab/MsdialWorkbench/files/12358102/MSDIAL.5.problem.pptx)

kozo2, 2023-08-22:
We reproduced the issue, and we fixed it at https://github.com/systemsomicslab/MsdialWorkbench/commit/838006d75ac6875cdafa02d0ed8868f775199e8b .
However, we found some additional issues that we have to address to correctly annotate lipids in EAD mode.
We will release the new version and let you know when it's ready.

Cagakan31, 2023-08-24:
Thank you @kozo2