systemsomicslab/MsdialWorkbench #182: Total Score Interpretation

vasiloa2, 2023-08-17:
Dear MS-DIAL developers, 

I would need some of your help to understand the 'total score' in the exported alignment result. 
I read the tutorial and the publication about the equations and how the total similarity is calculated with all four scores. 
In my exported files (please see an example attached) I see the 'RT similarity', 'm/z similarity' and the dot and reverse dot products used for the MSMS similarity but I do not see the isotope ratio similarity column. Why is that please? 
Another thing I am struggling with, is how to interpret the total score column. What would be the 'perfect' similarity score when using all four scores, like in this example here? 
I get, for example, in the total score column a value of 1.64, how can I interpret this, if this is a high score or not?
[exported alignment result.xlsx](https://github.com/systemsomicslab/MsdialWorkbench/files/12368378/exported.alignment.result.xlsx)

Thank you and best regards
Katerina

YukiMatsuzawa, 2023-08-24:
Hello @vasiloa2 ,

In the current version of MSDIAL, the Total Score value is derived from the weighted sum of each individual similarity score. For instance, within LCMS projects, similarity metrics such as retention time similarity, precursor m/z similarity, simple dot product, weighted dot product, reverse dot product, and matched peaks percentage. The weight parameter for these metrics is determined internally within MSDIAL for each project configuration.

Regarding the absence of isotope similarity values in the output, it seems that although these values are calculated internally, they are not currently being displayed. However,  the current version of MSDIAL, the isotope similarity value does not impact the calculation of the total score.

Best regards,
Yuki

olyabukato, 2023-08-24:
Dear @YukiMatsuzawa 
Thaks for You answer. Actually I have the same question as @vasiloa2. Now I basically understand how calculated "Match Scor" but I use identification w/o alignment and there is no match point in RT. In this case Total Score does not use this parameter? 
And I also interestind how can I find some trust score or cut off score trustworthy score? Like FDR for proteomics research.

vasiloa2, 2023-08-28:
Dear @YukiMatsuzawa, 

thank you for your reply, this is very helpful. I would only have one more question to help me understand the total score value. If we assume that the metrics you mentioned above - retention time similarity, precursor m/z similarity, simple dot product, weighted dot product, reverse dot product, and matched peaks percentage - have individually the perfect score, what would be the total score then?
This will help me to evaluate the total score of my experiment e.g. I get a value 1.8 out of what?

My best regards
Katerina