systemsomicslab/MsdialWorkbench #206: MSDIAL GUI does nothing after running new project

benton-anderson, 2023-09-08:
version 5.1.230807

The important project parameters are SWATH, lipidomics with .MSP library.

What do you need from me to replicate the problem?

kozo2, 2023-09-11:
Hi @benton-anderson 
Could you use the [MSDIAL5_paramexport.zip](https://github.com/systemsomicslab/MsdialWorkbench/releases/download/issue206/MSDIAL5_paramexport.zip) attached to https://github.com/systemsomicslab/MsdialWorkbench/releases/tag/issue206 to export the parameters.
And then reply with that text file here?
The parameter export function is not available in Version 5.1.230807.
(So you'll need to use the MSDIAL5_paramexport.zip I mentioned earlier.)

Knowing the details of the parameters will potentially allow us to provide a better response.
(That's why I'm asking for this. 😄 )
Thank you in advance 🙏 .

kozo2, 2023-09-11:
@benton-anderson 
You can export your parameters by clicking this button. (with the [MSDIAL5_paramexport.zip](https://github.com/systemsomicslab/MsdialWorkbench/releases/download/issue206/MSDIAL5_paramexport.zip))
[image]

closed 2025-02-12 by Bujee415