systemsomicslab/MsdialWorkbench #215: Problem when entering data, I do not see the alignments with my Msp database and they are not represented well in the bar graph

RocioMartin1, 2023-09-27:
[image]
Hello, 

I have entered the data in abf format, and the database in msp. But after loading everything and configuring the parameters I don't see any alignment with my database. Also, as you see the bar graph does not look normal, could someone help me? I cant work with this. 

Thank you

RocioMartin1, 2023-09-27:
[Primera prueba tratamiento de datos MSDial_26092023.docx](https://github.com/systemsomicslab/MsdialWorkbench/files/12736658/Primera.prueba.tratamiento.de.datos.MSDial_26092023.docx)

Bujee415, 2024-12-20:
Hello!
Sorry for the late reply!
Did you solve this problem? Which version of MS-DIAL is used here?
I highly recommend retrying with the newest version of MS-DIAL.
Thanks!