systemsomicslab/MsdialWorkbench #241: Software Instability with Large Sample Size Untargeted Metabolomics

DarioS, 2023-10-17:
Using the graphical version on Windows 11, the program window disappears without an error dialog. Sample size is close to 1000.

YukiMatsuzawa, 2024-02-19:
I apologize for the delayed response. I have a few questions to ask you. Which version of MSDIAL are you using? At which point in the analysis does the crash occur? What type of analysis were you performing (LCMS, GCMS, etc.)?

We have confirmed that it works on Windows 11 in our environment. However, the issue might be related to the large number of files. If that's the case, the latest version of MSDIAL has attempted to address this issue. Please give it a try.

wzhang15, 2024-02-28:
I have the same problem. It just shutdown wothout anything when I click the "compound search" button. My version is 4.9, in windows with around 30 data.
When I want to try the newest version or older version, the alignment data formate is not avaiable, that's a little bit inconvenient. Should I re analysis these data?