systemsomicslab/MsdialWorkbench #253: Questions about MSDial in command line

Ellakai, 2023-11-01:
Hello,

Thanks for your help and your great software :)

I have 2 projects with MSDial ; and for each project, i have 1 "issue" :(

1st project :
*************
This project is based on Windows app. I would like to performed my GCMS treatment with MSDial (4.8) in command line, for optimize my results. For that, i need to have the possibiliy to export the aligned file (Area and Height).
How i can do that ? Actually i just obtain 1 .msdial file per sample. In linux version, i obtain this files, but not in windows.

2nd project :
**************
This project is based on Linux app. For routine treatment, i have one linux server. All data treatment is perfect.
But, when i increase the number of thread, nothing change.
How i can increase that ?

Thanks again for your help,
regards

Ellakai, 2023-11-08:
Hello,
Sorry to bump my post, but I'm currently stuck. I imagine it's just two little things.
Thank you.

Bujee415, 2024-12-20:
Hello!
I deeply apologize for the late reply!
Did you solve this problem? Which version of MS-DIAL is used here?
Thanks!

closed 2025-02-12 by Bujee415