systemsomicslab/MsdialWorkbench #319: extracting feature list created from ms/ms fragment ion searcher

shota-matsumura, 2024-02-27:
Hi, 

I was wondering if there is a way to export just the MS features contained within a given ms/ms fragment search? If not, is there an alternative approach within the msdial software by which i can mark the compounds found within the ms/ms fragment search prior to exporting the feature list? 

I am currently using msdial v 5.1 .23

Thank you,

Warm regards,

Shota.

shota-matsumura, 2024-02-27:
Also, 

Thanks in advance for providing such a great open source platform to the MS community!

YukiMatsuzawa, 2024-04-04:
I apologize for the delayed response.

In MSDIAL, when exporting peak spotting results, you can use the current filtering parameters to narrow down the peaks that will be exported. The way to do this is by applying your filters in the main view or table view, then proceed to Export and check the 'Filtering by current parameter' option before executing the export. However, please note that this feature is currently available only after performing Alignment.

I hope this helps with your query.

[image]

closed 2024-12-20 by Bujee415