systemsomicslab/MsdialWorkbench #335: Changing LBM when uploading new experiment

acbarne5, 2024-03-29:
Hello, 

I am trying to change the lipid database to the Fiehn Lab database instead of the default, and when I do, it crashes the program. I have downloaded the new LBM2 file, put it into the MS DIAL folder, and deleted the default LBM2. Is there a step I am missing that causes the crash? I tried this with versions 5.1.230912 and 5.2.40328.alpha and had the same problem with both versions. 

Thank you for any help you can provide!

YukiMatsuzawa, 2024-04-04:
Hello,

I apologize for the lack of explanation on this matter. The Lbm files used in MSDIAL4 and MSDIAL5 have actually become entirely different entities. Therefore, the Lbm files distributed on the CompMs page can only be used with MSDIAL4.

Currently, the only Lbm files that can be used with MSDIAL5 are those that come bundled with the software.

Yuki

acbarne5, 2024-04-04:
Thanks, Yuki. I appreciate your response

So if I want to use the Fiehn Lab's retention times, is there a way to do that in MSDIAL5? Do I just have to select it somewhere?

YukiMatsuzawa, 2024-04-05:
Hmm, currently, the Lbm file is becoming quite large, and we are considering removing the Lbm file at some point, so we do not plan to distribute other Lbm files at the moment. We are thinking about adding a feature to MSDIAL that would allow for the correction of retention times in Lbm, but nothing has been decided as of yet.

acbarne5, 2024-04-08:
I see. I am using MS Dial because I know that my RTs are very similar to the Fiehn Lab and I want to be able to use RT as an identification method without having to create all my own RT library

YukiMatsuzawa, 2024-04-23:
We have prepared an MSDIAL5 Lbm file with RT adjustments specifically for the Fiehn Lab. Please try replacing it with this one and see how it works for you.

https://wetransfer.com/downloads/a635d92329628cdf1460691944e022eb20240422235512/97e91875161b6308a444b5f7adee123d20240422235512/82c137

acbarne5, 2024-05-13:
I've been busy and did not get this downloaded. The link has expired. Could you post the link again? Thank you!

YukiMatsuzawa, 2024-05-14:
Here is the new one: https://we.tl/t-5nFyMFGjMH

acbarne5, 2024-05-22:
Do you have another way to send the file? It is not letting me download it. Maybe I am doing something wrong.