systemsomicslab/MsdialWorkbench #336: Blank subtraction not working

ajburx, 2024-04-02:
Hello, when using MSDIAL-v5.2.240218.2 in hard ionisation GC-MS analysis mode and setting blank subtraction parameters, no blank subtraction is applied. The filtering options in the peak spot navigator are greyed out. Changing the blank subtraction threshold or other options e.g. "keep reference matched metabolite features" does not help.

YukiMatsuzawa, 2024-04-04:
Hello,

Thank you for reporting this bug. In fact, the migration of some GCMS utilities to MSDIAL5 has not been completed. We will verify the current behavior and proceed with the necessary corrections.

YukiMatsuzawa, 2024-04-24:
We have released a version of MSDIAL that includes the functionality for GCMS blank filtering. Please download it from the release page and check how it works. Thank you!

closed 2024-07-24 by ajburx