systemsomicslab/MsdialWorkbench #339: Error in Joint Aligner

acbarne5, 2024-04-10:
I am trying to process large datasets using a batching method where I run pooled samples to create a retention time and compound library, and then I use that library for the subsequent files from my large experiment. However,  I keep having errors in Joint Aligner. It tells me that there are files that do not have peaks. The files change from time to time, though, and they are never the same files. My machine has 64 GB of RAM. I used the resource monitor while uploading my files, and my CPU usage was around 50% and over 100% max frequency. Memory, however, was only around 30% when I loaded 4 files at a time. This still gave me two failed upload errors where it said there was no peak information in the sample. I knew this was untrue because those samples had been fine on Friday when I tried a batch. I finally was able to get it to go through by bumping it down to uploading 2 files at a time, but then no MS2 reference spectra showed up in this batch. Is it still a memory issue? If so, I don't understand why the resource monitoring is not showing it going above ~35%.

Thanks for any insight you can provide! I really appreciate this forum!

YukiMatsuzawa, 2024-04-24:
It sounds like a frustrating situation. In reality, MSDIAL currently has some issues with handling large datasets and can behave unstably. Often, these problems are related to memory or aspects of file handling. Since it seems unlikely to be a memory issue in your case, it could be related to how files are being saved.

I would like to ask a few questions to better understand your situation:

Which version of MSDIAL are you using, 4 or 5?
If you are using MSDIAL 4, are you using the GUI or Console version?
Are MS/MS spectra acquired using DDA or DIA?
What is the format of your spectral data (raw, wiff, mzml, abf, etc.)?
Thank you for your cooperation. I look forward to your response!

acbarne5, 2024-05-13:
In this case I was using MSDIAL version 4.9 GUI. I converted my data to abf, but they are originally Thermo RAW files. And it is DDA UHPLC-MS/MS