systemsomicslab/MsdialWorkbench #357: open ver.4.9.221218 in macbook

ziyuuua, 2024-05-08:
I couldn't find the executable file after extracting the .zip file from the website

kozo2, 2024-05-09:
Hi @ziyuuua 
http://prime.psc.riken.jp/compms/msdial/download/repository/macOS/MSDIAL%20ver.4.9.221218%20OSXx64.zip
does not have GUI executable file.

`MsdialConsoleApp` is the only available executable file on Mac. (See the attached image.)

[image]

JRodencal, 2024-05-09:
Hello! I see the file MsdialConsoleApp, but its kind is listed as "document". When I attempt to open the file I just get a text document. I assume that there may be some other program required to open this file type on Mac. Are there any existing tutorials on how to do this? Is this a known issue?

JRodencal, 2024-05-09:
I realized that I wanted to work with the GUI, not the console, so I gave up and decided to use a Windows virtual machine. I'm just running everything in windows now and it seems to work ok.

kozo2, 2024-05-09:
@JRodencal 🙏

closed 2024-05-09 by kozo2