systemsomicslab/MsdialWorkbench #366: MSDIAL crashes

vgikas, 2024-05-24:
I am trying to analyze data from a TIMS TOF PRO2 and an Impact instrument. The last version that could handle bbcid data was 4.9. The new versions crash after the alignment procedure. The DDA mode on the contrary works like a charm. I am using an I9 pc with 64 GB RAM with Windows 11 pro.
Could you please check this issue?

Bujee415, 2024-12-20:
Hello!
Sorry for the late reply!
Did you solve this problem? Which version of MS-DIAL is used here?
Thanks!

vgikas, 2025-01-03:
Hello!
Thanks for answering! I was using the latest available version at that time. Still the 5.5 version crashes. This happens at the end of the alignment process It seems like it cant punch the alignment file to the disk, but the other procedure files are written as expected. This happens to the DIA processing. If I process the same files (bbCID using a Bruker impact or TIMS TOF Pro2) by DDA the procedure ends normally but of course with no MSMS spectra assigned to the corresponding EIC's. I hope this could help you. Please don't hesitate to ask me perform any new processing with a different configuration suggested by you. I would be glad to help.

Bujee415, 2025-01-07:
Thanks for the reply!
Can you send us some example files and your parameters so that we can reproduce your problem? That would be so helpful in solving this problem. If you want to send us those files by email, please use "msdial-jp-groups@go.tuat.ac.jp".

closed 2025-02-26 by Bujee415