systemsomicslab/MsdialWorkbench #374: MSDIAL-v5.3.240617 crashes when using MSP library for identification for GC-MS project

QizhiSu, 2024-06-19:
Hi msdial developers,

Thanks a lot for uploading the latest MSDIAL!  With this new version, I faced crashing issues in GC-MS project. When not selecting MSP library for identification, everything is fine. However, when applying msp library for identification, it crashes. This msp library we used works perfectly with previous version, including MSDIAL version 4+. Below is how the msp library looks like. Any idea?
[image]

Thanks in advanced.
Best regards, 
Sukis

ajburx, 2024-07-23:
Hello, I also have this problem - we are using a custom library .msp file which does work fine with MSDIAL v4+, but causes a crash/program close when selected with MSDIAL v5+. The custom library is quite large and 500Mb. Analysis completes successfully with no library file selected.

Bujee415, 2024-12-20:
Hello!
Sorry for the late reply!
Did you solve this problem? Which version of MS-DIAL is used here?
Thanks!

closed 2025-05-14 by Bujee415