systemsomicslab/MsdialWorkbench #377: Aligned EIC file cannot be found. Please execute peak alignment fo rhte use of aligned EIC viewer

YHFoodScience, 2024-06-26:
We are using MS-DIAL 4.9.221218 for processing GC-EI-MS data. It worked well until 1 month ago when we re-processed the data and the the software shows use the notice of "The aligned EIC file (AlignmentResults ....EIC.aef) cannot be found. Please execute peak alignment fo rthe use of aligned EIC viewer." (a screenshot is attached below) Since we start to see this notice, the window for the "EIC of aligned spot" doesn't show anything and the aligned spot viewer shows blue dots for all features, instead of color coding them as before.  (a screenshot is attached below) 

We are unsure about where the issue come from. 

[image]

[image]

Bujee415, 2025-05-26:
Hello.
I am not sure about this issue. I think there is no way that the version 4.9 has changed, so my guess is "...EIC.aef" file is deleted somehow. I assume you tried creating a new project, but this error happens repeatedly. Did you tried it with using other folder for the project?

YLadha231, 2026-01-16:
Hi, I am having the same issue and have tried starting again from scratch and using a new folder, but it still keeps happening.