systemsomicslab/MsdialWorkbench #411: MS-DIAL 5 parameter export file (.txt) does not match file type of parameter import (.mdparameter)

andrewstai, 2024-09-30:
Hello! When I am exporting the parameters of a MS-DIAL 5 run to use for a future run, it only allows the parameters to be exported as a .txt file. However, the "load parameter" option when setting up a new project's parameters only allows .mdparameter files to be used for import. Currently I cannot find a way to get a .mdparameter export file, but I may be missing something.

montmasis, 2024-10-04:
I was wondering the same! Seems like a functionality which has been lost from previous MSDIAL v4. The .mdparameter used to be the default export, and the readable txt file had to be exported separately via the alignment results / parameter export.

montmasis, 2024-10-04:
Found it, it seems available to export the parameter files from the main menu, "Save Parameter as"...

andrewstai, 2024-10-04:
Oh I found it now as well, thank you! I wouldn't have thought to look there, so I'm glad you were able to find it.

closed 2024-10-04 by andrewstai