systemsomicslab/MsdialWorkbench #412: Peptide identification and custom modifications

jiunyanding, 2024-10-01:
Hello,

I am currently using MS-DIAL5 for proteomics analysis in my research. Since my samples are yeast proteins, the glycosylation patterns differ from those of humans. As a result, there are no corresponding options in the list of "modification setting" on the identification page. I would like to ask if it is possible to manually add custom modifications, and if so, could you please provide the method for doing so? 

Thank you for your assistance!

Bujee415, 2024-12-25:
Hello!
Sorry for the late reply!
Did you solve this problem? I highly recommend retrying it with the newest version (if you are using Visual Studio to build MS-DIAL, try after pulling the newest changes).
Also, we are preparing a tutorial for proteomics, so please wait a while!
Thanks!

jiunyanding, 2024-12-30:
Dear Bujee415,

I have recently switched to using MSDIAL-v5.5.241113, but I noticed that in the modification field, the available options for O-glycosylation still do not include the yeast-specific model (S/T-nHex).

Regarding MSDIAL-console, I couldn’t find any user instructions. However, I noticed that one argument involves using a method file. Is it possible to define custom modifications, similar to MS-GF+ (https://github.com/MSGFPlus/msgfplus), to create a database and perform subsequent peptide matching?

Thank you very much for your assistance!

Bujee415, 2025-01-07:
Thanks for the reply!
I hope this tutorial helps with the console version! "https://systemsomicslab.github.io/msdial5tutorial/utils.html#console-app"
You can use your own library on both the console and GUI app to annotate/match (add your database by clicking "+" icon on the identification page).
Let me know if there is any misunderstanding or question that isn't answered!

Bujee415, 2025-01-20:
Hello again!
If you need, we uploaded a tutorial video on proteomics!
https://www.youtube.com/watch?v=kdykKiKPaho

Thank you!

closed 2025-01-20 by Bujee415