systemsomicslab/MsdialWorkbench #430: Molecular network export option - Scoring

Roninaro, 2024-10-23:
Hello everyone, 
which scoring algorithm (modified cosine score eg) is used when I use the molecular network export feature to export the nodes and edges table to load it into cytoscape? We had recently a discussion about scoring algorithms for FMNB and now are curious which one is used for MS-Dial. 
Thank you for your help, 
Roninaro

kozo2, 2024-10-23:
Hi Roninaro,
we use "Bonanza spectral clustering algorithm" [Falkner, J. A., Falkner, J. W., Yocum, A. K. & Andrews, P. C. J. Proteome Res. 7, 4614–4622 (2008)].

closed 2024-10-23 by Roninaro

judezhang1997, 2024-11-14:
Hi, the data of Agilent Qtof were imported into msdial 5,  and every compounds with 3 kinds of CEs, have 3 spectrums. So, how can i deal with the data into the molecular network，and let the melecular network more accuarate?