systemsomicslab/MsdialWorkbench #435: Cannot create project (MSDIAL v. 5.4.241021)

miko353, 2024-10-30:
Hello there. I recently downloaded MS-DIAL and when I try to create new project, I can only get to Measurement parameters, when I click next, the app crashes without any error message. I tried building it in Visual Studio, but the result is the same. I get a couple of exceptions and the app crashes. I tried it with my converted .raw to .abf data, and also with the test wine LC-MS/MS .abf dataset from your website.

I use Win 11 Pro, v. 23H2, 16GB, i7-1255U. However, the app works fine on other PCs in our lab with Win 10 or Win 11 Home (also v. 23H2, 16GB). I created test projects on these PCs and managed to open them on my PC. Any idea how to fix it?

Bujee415, 2024-12-06:
Hello! Sorry for the late reply!
Which version of Visual Studio used here? Also, if possible I suggest building it in Visual Studio now (we recently added error messages when anything happens during loading). 
Let me know if you succeed or not! Thanks!

miko353, 2024-12-10:
Hello, since then, I had to reinstall my Win, so I cannot tell you which version of Visual studio I used. However the newest version of MS Dial works fine now after clean install. I'm not sure what could have been the issue there.

Bujee415, 2024-12-10:
It's nice to hear it is working without any problem now!
Have a great week!

closed 2024-12-10 by Bujee415