systemsomicslab/MsdialWorkbench #438: Urgent: Recurrent Task Interruptions and Memory Overload Issues with MSDIAL.ver5.3.240719 During Metabolomics Data Analysis

liaochenlanruo, 2024-11-05:
Dear Developer Team,

Hello! I am encountering some technical challenges while using MSDIAL.ver5.3.240719 for metabolomics data analysis and would greatly appreciate your assistance and support. The specifics are as follows:

- I am currently working with a dataset that includes 889 samples, collected via Waters UPLC-Qtof in MSE mode. Each sample file size ranges from 500MB to 2.5GB.
- During the processes of peak detection, identification, and alignment, tasks frequently interrupt unexpectedly, preventing the completion of the entire analytical workflow. When this occurs, the system displays an error message similar to the one shown below. For the data that fails to process successfully, the expected output files—sample.pai2, sample_tags.xml, and sample.dcl—are typically not generated. If I manually remove the problematic data and attempt to rerun the program, new problematic data points usually emerge.
- Additionally, after completing the aforementioned steps for all samples individually, when entering the final stage of global alignment, my computer experiences severe lag or even becomes completely unresponsive. At this time, checking the resource monitor reveals that the memory usage approaches or exceeds 95% (my system has 128GB RAM).

[image]

Given these circumstances, I kindly request your team’s assistance in investigating the root causes of these issues and considering providing corresponding solutions or optimization suggestions to enable large-scale data analysis. Your support and attention are highly appreciated!

Best regards,
Hualin Liu

shiyuxinppppp, 2024-11-12:
你好，我想问一下，你使用的MSe模式raw有使用格式转换吗，例如mzml等，如果有，你使用的转换软件是什么？

liaochenlanruo, 2024-11-12:
> 你好，我想问一下，你使用的MSe模式raw有使用格式转换吗，例如mzml等，如果有，你使用的转换软件是什么？

使用了msdial官网的转换软件，ABF file converter，转换为abf格式

shiyuxinppppp, 2024-11-12:
谢谢，另外我想问一下，使用abf转换时候，mse数据应该勾选哪些设置是对的，因为我转换后的数据没有二级谱图，导致无法与数据库匹配。
[image]

shiyuxinppppp, 2024-11-12:
不好意思，我还想问一下，就是在导入数据部分，您选择的采集类型是AIF吗
[image]

Catosine, 2024-11-19:
Hello all,

I had a similar situation as you described. Based on my observations, it seems to be a problem with multithreading.
Maybe you could try using v5.4 with no multithreading?
I am working on a GCMS dataset with 3k+ samples. And I tried MSDial 4.9 and 5.4.

Good luck!

Bujee415, 2024-12-18:
Hello!
Sorry for the late reply!
I suggest using new version of MS-DIAL and if there is still a problem, let me know.
(If you using Visual Studio to build MS-DIAL, I highly recommend retry after pulling!)

closed 2025-04-11 by Bujee415