systemsomicslab/MsdialWorkbench #457: 5.5.241113 crashes when trying to open  wiff2 files

tiwaho, 2024-11-19:
We tried in a normal way to open sciex wiff2 files and program just shuts down.

tiwaho, 2024-12-03:
We have a similar issue with the wiff2 files. The program runs into an exception when loading the data. The error is below.\

Could anyone help?

```
System.AggregateException
  HResult=0x80131500
  Message=One or more errors occurred.
  Source=mscorlib
  StackTrace:
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at CompMs.MsdialCore.Algorithm.BaseDataProvider.LoadMs1Spectrums() in C:\Users\Empower3\Source\Repos\MsdialWorkbench\src\MSDIAL5\MsdialCore\Algorithm\BaseDataProvider.cs:line 84
   at CompMs.MsdialCore.Algorithm.DataProvider.GetMs1Range(IDataProvider provider, IonMode ionMode) in C:\Users\Empower3\Source\Repos\MsdialWorkbench\src\MSDIAL5\MsdialCore\Algorithm\IDataProvider.cs:line 32
   at CompMs.MsdialCore.Algorithm.PeakSpottingCore.Execute3DFeatureDetectionNormalModeByMultiThread(AnalysisFileBean file, IDataProvider provider, Int32 numThreads, CancellationToken token, ReportProgress reporter, ChromatogramRange chromatogramRange) in C:\Users\Empower3\Source\Repos\MsdialWorkbench\src\MSDIAL5\MsdialCore\Algorithm\PeakSpottingCore.cs:line 112
   at CompMs.MsdialCore.Algorithm.PeakSpottingCore.Execute3DFeatureDetectionNormalMode(AnalysisFileBean file, IDataProvider provider, Int32 numThreads, CancellationToken token, ReportProgress reporter, ChromatogramRange chromatogramRange) in C:\Users\Empower3\Source\Repos\MsdialWorkbench\src\MSDIAL5\MsdialCore\Algorithm\PeakSpottingCore.cs:line 52
   at CompMs.MsdialCore.Algorithm.PeakSpottingCore.Execute3DFeatureDetection(AnalysisFileBean file, IDataProvider provider, Int32 numThreads, CancellationToken token, ReportProgress reporter, ChromatogramRange chromatogramRange) in C:\Users\Empower3\Source\Repos\MsdialWorkbench\src\MSDIAL5\MsdialCore\Algorithm\PeakSpottingCore.cs:line 35
   at CompMs.MsdialLcMsApi.Algorithm.PeakSpotting.Run(IDataProvider provider, MsdialLcmsParameter param, IProgress`1 progress, CancellationToken token) in C:\Users\Empower3\Source\Repos\MsdialWorkbench\src\MSDIAL5\MsdialLcMsApi\Algorithm\PeakSpotting.cs:line 28
   at CompMs.MsdialLcMsApi.Process.PeakPickProcess.Pick(AnalysisFileBean file, IDataProvider provider, IProgress`1 progress, CancellationToken token) in C:\Users\Empower3\Source\Repos\MsdialWorkbench\src\MSDIAL5\MsdialLcMsApi\Process\PeakPickProcess.cs:line 19
   at CompMs.MsdialLcMsApi.Process.FileProcess.FindPeakAndScans(AnalysisFileBean analysisFile, IDataProvider provider, IProgress`1 progress, CancellationToken token) in C:\Users\Empower3\Source\Repos\MsdialWorkbench\src\MSDIAL5\MsdialLcMsApi\Process\FileProcess.cs:line 78
   at CompMs.MsdialLcMsApi.Process.FileProcess.<RunAsync>d__5.MoveNext() in C:\Users\Empower3\Source\Repos\MsdialWorkbench\src\MSDIAL5\MsdialLcMsApi\Process\FileProcess.cs:line 47
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.ConfiguredTaskAwaitable.ConfiguredTaskAwaiter.GetResult()
   at CompMs.MsdialCore.Algorithm.ProcessRunner.Consumer.<ConsumeAsync>d__5.MoveNext() in C:\Users\Empower3\Source\Repos\MsdialWorkbench\src\MSDIAL5\MsdialCore\Algorithm\ProcessRunner.cs:line 36
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.ConfiguredTaskAwaitable.ConfiguredTaskAwaiter.GetResult()
   at CompMs.App.Msdial.View.Core.MainWindow.<>c__DisplayClass7_2.<<ShowMultiProgressBarWindow>b__0>d.MoveNext() in C:\Users\Empower3\Source\Repos\MsdialWorkbench\src\MSDIAL5\MsdialGuiApp\View\Core\MainWindow.xaml.cs:line 247
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   at System.Windows.Threading.DispatcherOperation.InvokeImpl()
   at MS.Internal.CulturePreservingExecutionContext.CallbackWrapper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at MS.Internal.CulturePreservingExecutionContext.Run(CulturePreservingExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Windows.Threading.DispatcherOperation.Invoke()
   at System.Windows.Threading.Dispatcher.ProcessQueue()
   at System.Windows.Threading.Dispatcher.WndProcHook(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndWrapper.WndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndSubclass.DispatcherCallbackOperation(Object o)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
[... cut ...]
    CompMs.MsdialCore.Algorithm.BaseDataProvider.LoadMeasurementAsync(CompMs.MsdialCore.DataObj.AnalysisFileBean, bool, bool, bool, int, System.Threading.CancellationToken) in BaseDataProvider.cs
    CompMs.MsdialCore.Algorithm.BaseDataProvider..ctor.AnonymousMethod__0() in BaseDataProvider.cs
    CompMs.MsdialCore.Algorithm.BaseDataProvider.LoadMsNSpectrumsAsync.AnonymousMethod__1() in BaseDataProvider.cs

Inner Exception 1:
FileLoadException: Loading Z:\RawData\Sciex training days\yeast 1x dil. 2 CG 35.wiff2 failed.
```

tiwaho, 2024-12-05:
The following errors are also probably related to this.

```
Severity	Code	Description	Project	File	Line	Suppression State	Details
Error (active)	CS0246	The type or namespace name 'RawDataAccess' could not be found (are you missing a using directive or an assembly reference?)	MsdialCore (net48), MsdialCore (net6.0), MsdialCore (netstandard2.0)	C:\Users\Empower3\source\repos\MsdialWorkbench\src\MSDIAL5\MsdialCore\DataObj\AnalysisFile.cs	133		
Error (active)	CS0246	The type or namespace name 'RawDataAccess' could not be found (are you missing a using directive or an assembly reference?)	MsdialCore (net48), MsdialCore (net6.0), MsdialCore (netstandard2.0)	C:\Users\Empower3\source\repos\MsdialWorkbench\src\MSDIAL5\MsdialCore\DataObj\AnalysisFile.cs	138		
Error (active)	CS0246	The type or namespace name 'RawDataAccess' could not be found (are you missing a using directive or an assembly reference?)	MsdialCore (net48), MsdialCore (net6.0), MsdialCore (netstandard2.0)	C:\Users\Empower3\source\repos\MsdialWorkbench\src\MSDIAL5\MsdialCore\Utility\DataAccess.cs	54		
Error (active)	CS0246	The type or namespace name 'RawDataAccess' could not be found (are you missing a using directive or an assembly reference?)	MsdialCore (net48), MsdialCore (net6.0), MsdialCore (netstandard2.0)	C:\Users\Empower3\source\repos\MsdialWorkbench\src\MSDIAL5\MsdialCore\Utility\DataAccess.cs	66		
Error (active)	CS0246	The type or namespace name 'RawDataAccess' could not be found (are you missing a using directive or an assembly reference?)	MsdialCore (net48), MsdialCore (net6.0), MsdialCore (netstandard2.0)	C:\Users\Empower3\source\repos\MsdialWorkbench\src\MSDIAL5\MsdialCore\Utility\DataAccess.cs	76		
Error (active)	CS0246	The type or namespace name 'RawDataAccess' could not be found (are you missing a using directive or an assembly reference?)	MsdialCore (net48), MsdialCore (net6.0), MsdialCore (netstandard2.0)	C:\Users\Empower3\source\repos\MsdialWorkbench\src\MSDIAL5\MsdialCore\Utility\DataAccess.cs	91		

```

Bujee415, 2024-12-18:
Hello!
Sorry for the late reply!
Did you try it with wiff file? And which version of MS-DIAL are you using?
If you using Visual Studio to build MS-DIAL, I highly recommend retry after pulling!

tiwaho, 2024-12-18:
Thank you for replying! I have now tried with both wiff and wiff2 files.. no luck. We have indeed been using visual studio to build the program. The version we are trying to get working is the latest 5.5.24113 version.  When I try the pulling, visual studio says that the repository is already up to date? Running the program in visual studio results in crashing all the time and gives the error message as in the attachment..
[image]

tiwaho, 2024-12-18:
> Hello! Sorry for the late reply! Did you try it with wiff file? And which version of MS-DIAL are you using? If you using Visual Studio to build MS-DIAL, I highly recommend retry after pulling!
Thank you for replying! I have now tried with both wiff and wiff2 files.. no luck. We have indeed been using visual studio to build the program. The version we are trying to get working is the latest 5.5.24113 version. When I try the pulling, visual studio says that the repository is already up to date? Running the program in visual studio results in crashing all the time and gives the error message as in the attachment..
[image]

Bujee415, 2024-12-19:
Sorry, I couldn't reproduce this error.
Could you send us some example data?
Just in case, if using email is easy for you, use this email to contact us "msdial-jp-groups@go.tuat.ac.jp".
Thanks!

tiwaho, 2024-12-20:
Hi, yes I can send you an example datafile that was recorded. Perhaps you get the same error messages. I will email it to the emailaddress you provided me.

Bujee415, 2024-12-20:
Hello, I just sent you an email.
Please check your file extension!
Thanks!