systemsomicslab/MsdialWorkbench #461: MSDIAL auto shut down issue

Sehrawatsurender, 2024-11-25:
Hi
I am trying to use proteomics workflow in MSDIAL. 
The software automatically shuts down when we start run.
I tried several times.

Files
I have .RAW files from Thermo orbitrap.
Although, .RAW can be directly accessed in MSDIAL, Do I need to change the file format?
Please provide suggestions.

Bujee415, 2024-12-19:
Hello! 
Sorry for the late reply! Did you solve this problem?
Which version of MS-DIAL are you using?
Also if you using Visual Studio to build MS-DIAL, I highly recommend retrying after pulling the newest changes!

Bujee415, 2025-05-14:
Hello
It can be related to the language setting or windows version, but not that sure about it.
The newest of MS-DIAL shows an error window which is a little more readable, so if possible, can you try it with the newest version?
Thank you.

closed 2025-05-21 by Bujee415