systemsomicslab/MsdialWorkbench #478: How to run MSDIAL v5.5.241113 on Mac

Midreancgle, 2024-12-18:
Hello！

In the unzipped folder, I didn't find the MsdialConsoleApp. How do I run MS-Dial on my Mac？

Bujee415, 2024-12-19:
Hello!
See below and I am sure there is "MSDIALCUI.exe" in your unzipped folder.
https://systemsomicslab.github.io/msdial5tutorial/utils.html#console-app

closed 2024-12-19 by Bujee415

Bujee415, 2024-12-19:
Oh sorry for the misunderstanding! 
It is "MSDIALCUI" and it should work the same way.

Midreancgle, 2024-12-19:
Thanks!

After giving MSDIALCUI execution permission, it is ready to run. But I have a new problem. Can you help me solve it? The following information is displayed.

`$ ./MSDIALCUI lcms -i /Users/ryue/Downloads/MSDIALCUI_demo_lcms_lipidomics/rawdata -o /Users/ryue/Downloads/MSDIALCUI_demo_lcms_lipidomics/rawdata -m /Users/ryue/Downloads/MSDIALCUI_demo_lcms_lipidomics/lib/msdial_console_param4lipidomics.txt -p
Loading analysis files..
Start processing..
System.IO.FileNotFoundException: Could not load file or assembly 'RawDataHandler, Version=1.2.9013.327, Culture=neutral, PublicKeyToken=null'. The system cannot find the file specified.

File name: 'RawDataHandler, Version=1.2.9013.327, Culture=neutral, PublicKeyToken=null'
   at CompMs.MsdialCore.Algorithm.BaseDataProvider.LoadMeasurementAsync(AnalysisFileBean file, Boolean isProfile, Boolean isImagingMs, Boolean isGuiProcess, Int32 retry, CancellationToken token)
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at CompMs.MsdialCore.Algorithm.BaseDataProvider.LoadMeasurementAsync(AnalysisFileBean file, Boolean isProfile, Boolean isImagingMs, Boolean isGuiProcess, Int32 retry, CancellationToken token)
   at CompMs.MsdialCore.Algorithm.StandardDataProvider..ctor(AnalysisFileBean file, Boolean isImagingMs, Boolean isGuiProcess, Int32 retry, CancellationToken token)
   at CompMs.MsdialCore.Algorithm.StandardDataProviderFactory.Create(AnalysisFileBean file)
   at CompMs.MsdialLcMsApi.Process.FileProcess.RunAsync(AnalysisFileBean analysisFile, ProcessOption option, IProgress`1 progress, CancellationToken token)
   at CompMs.MsdialCore.Algorithm.ProcessRunner.Consumer.ConsumeAsync(ProcessOption option, CancellationToken token)
   at CompMs.App.MsdialConsole.Process.LcmsProcess.ExecuteAsync(IMsdialDataStorage`1 storage, String outputFolder, Boolean isProjectSaved) -- One or more errors occurred. (Could not load file or assembly 'RawDataHandler, Version=1.2.9013.327, Culture=neutral, PublicKeyToken=null'. The system cannot find the file specified.
) --    at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at CompMs.App.MsdialConsole.Process.LcmsProcess.Run(String inputFolder, String outputFolder, String methodFile, Boolean isProjectSaved, Single targetMz)
   at CompMs.App.MsdialConsole.Process.MainProcess.Run(String[] args)`

ddomingof, 2024-12-23:
I am having the same problem, were you able to fix it?

Midreancgle, 2024-12-30:
Hello @Bujee415 :

Can you help me and @ddomingof  solve the above operational problems?

Look forward to your reply and resolution as soon as possible.

Thanks！

Bujee415, 2025-01-07:
Thanks for the reply!
Can you send us some example files so that we can reproduce this problem and solve this? (if you prefer sending us by an email, use "msdial-jp-groups@go.tuat.ac.jp").

JuanGonzalezm123, 2025-04-15:
Hello,

I'd like to know if you've been able to use MS-Dial on your Mac.
I've been trying and haven't been able to, so I appreciate your help.

Bujee415, 2025-04-15:
Hello, 

You can use console version of MS-DIAL on Mac!
Please refer the link below.
https://systemsomicslab.github.io/msdial5tutorial/utils.html#console-app

Thank you.