systemsomicslab/MsdialWorkbench #481: MS DIAL 5.5 Crushes during alingment

JohnFu4, 2024-12-21:
Hello,

I am experiencing an issue where MS-DIAL 5.5 (and other versions 5.X) consistently crashes when I try to perform data alignment (These data work fine in MS-DIAL 4.9). Although, in newer MS-DIAL version, I am evaluating them in proteomics mode, negative ion mode, with data measured in full scan mode on an LC-qTOF. The measured data are imported in mzML format.

In the Windows Event Viewer, I found the following report:

<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
- <System>
  <Provider Name=".NET Runtime" /> 
  <EventID Qualifiers="0">1026</EventID> 
  <Level>2</Level> 
  <Task>0</Task> 
  <Keywords>0x80000000000000</Keywords> 
  <TimeCreated SystemTime="2024-12-21T11:08:17.859759000Z" /> 
  <EventRecordID>1404124</EventRecordID> 
  <Channel>Application</Channel> 
  <Computer>CZBLKSYNTH660.cz.corporate.synthon-group.com</Computer> 
  <Security /> 
  </System>
- <EventData>
  <Data>Application: MSDIAL.exe Framework Version: v4.0.30319 Description: The process was terminated due to an unhandled exception. Exception Info: System.ArgumentNullException at CompMs.MsdialCore.Parser.MsdialProteomicsSerializer.LoadProteinResultContainer(System.String) at CompMs.App.Msdial.Model.Lcms.LcmsAlignmentModel..ctor(CompMs.App.Msdial.Model.DataObj.AlignmentFileBeanModel, CompMs.MsdialCore.Algorithm.Annotation.IMatchResultEvaluator`1<CompMs.Common.DataObj.Result.MsScanMatchResult>, CompMs.MsdialCore.DataObj.DataBaseStorage, CompMs.App.Msdial.Model.Search.PeakSpotFiltering`1<CompMs.App.Msdial.Model.DataObj.AlignmentSpotPropertyModel>, CompMs.App.Msdial.Model.Search.PeakFilterModel, CompMs.MsdialCore.DataObj.DataBaseMapper, CompMs.MsdialLcmsApi.Parameter.MsdialLcmsParameter, CompMs.App.Msdial.Model.DataObj.FilePropertiesModel, System.Collections.Generic.List`1<CompMs.MsdialCore.DataObj.AnalysisFileBean>, CompMs.App.Msdial.Model.DataObj.AnalysisFileBeanModelCollection, Reactive.Bindings.Notifiers.IMessageBroker) at CompMs.App.Msdial.Model.Lcms.LcmsMethodModel.LoadAlignmentFileCore(CompMs.App.Msdial.Model.DataObj.AlignmentFileBeanModel) at CompMs.App.Msdial.Model.Core.MethodModelBase.LoadAlignmentFileAsync(CompMs.App.Msdial.Model.DataObj.AlignmentFileBeanModel, System.Threading.CancellationToken) at CompMs.App.Msdial.ViewModel.Lcms.LcmsMethodViewModel.LoadAlignmentFileCoreAsync(CompMs.App.Msdial.ViewModel.DataObj.AlignmentFileBeanViewModel, System.Threading.CancellationToken) at CompMs.App.Msdial.ViewModel.Core.MethodViewModel.LoadAlignmentFileAsync(System.Threading.CancellationToken) at CompMs.App.Msdial.ViewModel.Core.MethodViewModel+<>c__DisplayClass0_0.<.ctor>b__11() at Reactive.Bindings.AsyncReactiveCommand+<>c__DisplayClass6_0+<<Subscribe>b__0>d.MoveNext() at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(System.Threading.Tasks.Task) at Reactive.Bindings.AsyncReactiveCommand`1+<ExecuteAsync>d__16[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].MoveNext() at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(System.Threading.Tasks.Task) at Reactive.Bindings.AsyncReactiveCommand`1+<Execute>d__15[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].MoveNext() at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() at System.Windows.Threading.ExceptionWrapper.InternalRealCall(System.Delegate, System.Object, Int32) at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(System.Object, System.Delegate, System.Object, Int32, System.Delegate) at System.Windows.Threading.DispatcherOperation.InvokeImpl() at MS.Internal.CulturePreservingExecutionContext.CallbackWrapper(System.Object) at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean) at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean) at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object) at MS.Internal.CulturePreservingExecutionContext.Run(MS.Internal.CulturePreservingExecutionContext, System.Threading.ContextCallback, System.Object) at System.Windows.Threading.DispatcherOperation.Invoke() at System.Windows.Threading.Dispatcher.ProcessQueue() at System.Windows.Threading.Dispatcher.WndProcHook(IntPtr, Int32, IntPtr, IntPtr, Boolean ByRef) at MS.Win32.HwndWrapper.WndProc(IntPtr, Int32, IntPtr, IntPtr, Boolean ByRef) at MS.Win32.HwndSubclass.DispatcherCallbackOperation(System.Object) at System.Windows.Threading.ExceptionWrapper.InternalRealCall(System.Delegate, System.Object, Int32) at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(System.Object, System.Delegate, System.Object, Int32, System.Delegate) at System.Windows.Threading.Dispatcher.LegacyInvokeImpl(System.Windows.Threading.DispatcherPriority, System.TimeSpan, System.Delegate, System.Object, Int32) at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr, Int32, IntPtr, IntPtr) at MS.Win32.UnsafeNativeMethods.DispatchMessage(System.Windows.Interop.MSG ByRef) at System.Windows.Threading.Dispatcher.PushFrameImpl(System.Windows.Threading.DispatcherFrame) at System.Windows.Application.RunDispatcher(System.Object) at System.Windows.Application.RunInternal(System.Windows.Window) at CompMs.App.Msdial.App.Main()</Data> 
  </EventData>
  </Event>

Thank you very much for your assistance.

Bujee415, 2025-02-26:
Hello!
Can you try it with the newest version?
Thank you!

JohnFu4, 2025-02-26:
Hello, I have tried the newest MS DIAL version, I was able to process data. Although when I performed double-click to open/activate the processed alignment, the following error appeared. 

[image]

Thank you for your effort and time.

Bujee415, 2025-02-26:
Hello!
I am not sure about this issue.
Can you send us the demo data and your analysis parameters so that we can reproduce your problem and solve this bug? If you prefer to send us by email, use msdial-jp-groups@go.tuat.ac.jp. 
Thank you!

closed 2025-05-21 by Bujee415