systemsomicslab/MsdialWorkbench #492: Smoothing method

FalcoB294, 2025-01-22:
### Discussed in https://github.com/systemsomicslab/MsdialWorkbench/discussions/440

<div type='discussions-op-text'>

<sup>Originally posted by **FalcoB294** November  8, 2024</sup>
Hi,

I have a question to the smoothing method than can be selected in MS-Dial.

Currently, we are using the Savitzky-Golay filter, as we are also doing when using XCMS.
In the MS-Dial tutorial the given method is the linear-weighted moving average method.
Is this a kind of general recommendation from MS-Dial?

Thanks!
FalcoB</div>

Bujee415, 2025-02-14:
Hello, there!
The default method is the linear-weighted moving average method, but if the other method meets your needs, it is free to use another smoothing method!
Does this answer your question?

FalcoB294, 2025-02-19:
Hi,

have you or anyone else any experience, what works better in MS-Dial: linear-weighted moving average vs. Savitzky-Golay? I read that integration is maybe better to use linear-weighted moving average method, because one can accurately determine the peak left- and right edges more precisely?

Best regards
FalcoB

Bujee415, 2025-04-15:
Hi!
Sorry for the late reply!
For our data analysis, we use linear-weighted moving average mainly and just added Savitzky-Golay as an option.
Please feel free to use the one you like (If you don't know which one, I recommend linear-weighted moving average. Also I don't think there is much difference).