systemsomicslab/MsdialWorkbench #513: Pathway data visualization MSDIAL

Kheme1a, 2025-02-21:
the pathway tool appears disabled, how can I make it work? how should I load the information? I am analyzing LC-ESI-QTOF samples in wiff2 format.
should i use any specific library? thanks

[image]

mxlinh2910, 2025-04-09:
Has there been an update on this issue yet? I'm facing the same problem. Could you please let me know if you solved it? Thank you in advance.

Bujee415, 2025-08-22:
Hello,

We apologize for the inconvienence. This function is currently under development. 
But if you want to use pathway mapping tool, you can export the MS-DIAL result and use it as input for other mapping tools. 
I recommend this tool https://github.com/systemsomicslab/MSLipidMapper and it can utilize MS-DIAL result directly. 
Hope it helps!

Best regards,
Bujee