systemsomicslab/MsdialWorkbench #533: MSDIAL alignement issue

vgikas, 2025-04-06:
Dear MSDIAL authors. It seems that the alignment is throwing some errors, could you plese look this up?

[image]

e look this up?

vgikas, 2025-04-06:
furthermore, the molecular spectrum networking pops out this issue.

[image]

no ms or ms/ms spectra can be found either the survey scan or the MSMS spectrum viewer

Can you please check those out too? I am using the April 4, 2025 version of MSDIAL and the demo wine data for a DIA analysis using the AIF feature

Bujee415, 2025-04-15:
Hello!
Can you send us any demo data so that we can reproduce the problem?
Please use this email if you like (msdial-jp-groups@go.tuat.ac.jp).
Thank you.

froz9, 2025-05-08:
I have the same issue, too.

vgikas, 2025-05-28:
Hello, 
We have used our datasets also, but we always "benchmark" the behavior of the software using the "All ion fragmentation with multiple collision energies" LC-MS/MS Dataset that is deposited to the main page of the systemsomicslab (https://systemsomicslab.github.io/compms/msdial/main.html). If more info is needed, we will be happy to share it.