systemsomicslab/MsdialWorkbench #556: Peak Curation (EIC overlay) Issue

Mcamp5678, 2025-05-06:
Dear MS-Dial Authors:

A previous user reported this error last month, but I wanted to follow-up to see if there was a solution. 

I am trying to re-integrate peaks using the "peak curation (EICs overlay)" and "Peak curation (Sample table)" component, but am seeing an empty pop-up. Is this a bug? or am I missing a key step in my data processing? 

Thanks!

[image]

Bujee415, 2025-05-14:
Hello,
We fixed this issue and are planning to release our new version.
So, please wait for a while for the release.
Thank you!

closed 2025-05-14 by Bujee415

Mcamp5678, 2025-06-16:
Hello,

When will the new version be released?