systemsomicslab/MsdialWorkbench #558: MsDial 5 error trying to read .mzml archives.

Vinil4, 2025-05-07:
I'm using MS-DIAL 5 version 5.5.250404  and I'm using  .mzml data archives, but when  finish all the setup of the app and start running the program process, appears the error window that I'm sending with this mail. I tried to run the app in three diferent computers but happens the same issue and I need this working as soon as possible.

If you could help me with this I will be very glad.

P.S.:  I already tried the standard version, the visial studio version and an old version too, but none of then it worked.

[image]

Bujee415, 2025-05-14:
Hello,
Can you send us a demo file so that we can reproduce the issue and fix it?
Please use this email for sending the data "[msdial-jp-groups@go.tuat.ac.jp](mailto:msdial-jp-groups@go.tuat.ac.jp)".
Thank you.

Vinil4, 2025-05-16:
I think its solved. Apparently my .mzml archives were without any information and with 0 bytes. So I did use the MsConvert to convert the .lcd archives into vendors and runned as well. I will test until the limit.

Bujee415, 2025-05-21:
Happy to hear that!
FYI, almost all files including .lcd can be directly imported into MS-DIAL.
If you have any other questions, let us know!

closed 2025-05-21 by Bujee415