systemsomicslab/MsdialWorkbench #559: MS-DIAL: "No peak information. Check your polarity setting." with Shimadzu LC-MS/MS data

Gio-Vi, 2025-05-07:
Hi everyone,

I’m working with LC-MS/MS data from a Shimadzu instrument, and I’m trying to analyze it using MS-DIAL. My raw files are in .lcd format. Here’s what I’ve done so far:
1. Converted .lcd files to .mzML using LabSolutions.
2. Converted .mzML files to .abf using Reifycs Analysis Base File Converter.
3. Imported both .mzML and .abf files into MS-DIAL and ran with the following settings:
4. MS1 data type: Centroid
5. MS/MS: Centroid
6. Ion mode: Positive
7. Project type: Metabolomics

However, I keep getting the message: “No peak information. Check your polarity setting.”
I’ve tried using all file types (.lcd, .mzML, and .abf), but nothing works.
Does anyone know what might be causing this issue?

Many thanks in advance for your support!
Best regards,

Gio

Bujee415, 2025-05-14:
Hello,
Can you send us a demo file so that we can reproduce the issue and fix it?
Please use this email for sending the data "msdial-jp-groups@go.tuat.ac.jp".
Thank you.

Gio-Vi, 2025-05-14:
Hello,

Many thanks for your answer.

Sure, I will send you an email with an archive containing:
1.	The original RAW file (.lcd)
2.	The converted .mzML in profile using the Shimadzu conversion tool
3.	The converted .mzML in centroid using the Shimadzu conversion tool

Let me know if you have any insight. Using SeeMS both MS1 and MS2 spectra appear to be correctly converted from profile to centroid mode. Additionally, I was able to visualize the converted spectra in MZmine.

The acquisition mode used was a _"segmented DDA"_ with different inclusion lists based on _m/z_ ranges. This approach was necessary due to acquisition limitations of the LC-MS/MS system.

Best regards,

Gio

sidra-azhar, 2025-05-15:
I m facing same issue #539 of "no peak information, Check polariity setting " Kindly tell me possible solution ?

Bujee415, 2025-05-21:
Hello,
Can you send us a demo file so that we can reproduce the issue and fix it?
Please use this email for sending the data "msdial-jp-groups@go.tuat.ac.jp".
Also, check if your data has MS1, because MS1 for peak picking (no MS1 can lead to no peak information).
Thank you.

sidra-azhar, 2025-05-21:
Yes...I have send demo file and it also include MS1 data..kindly check it
and tell me where problem lies ?

Bujee415, 2025-05-21:
Is the file name "Sample-B-MS-MS.raw"?

Bujee415, 2025-05-21:
Sorry for the issue.
I just assumed you will send us the raw data by an email...
If you prefer using github issue for the communication, we can definitely work here!

Gio-Vi, 2025-05-21:
It would probably be more useful for the community to have the discussion here.
Did you receive the files I shared with you

sidra-azhar, 2025-05-21:
Thanks #Bujee for your suggestion.

Bujee415, 2025-05-21:
Dear Gio-Vi,
We did not recieve any files...
Can you check it again?
Thank you.

Gio-Vi, 2025-05-21:
> Dear Gio-Vi, We did not recieve any files... Can you check it again? Thank you.

Just sent it — please check the msdial-jp-groups@go.tuat.ac.jp inbox and also the junk folder.

Bujee415, 2025-05-22:
We got it, and I checked your raw data. It seems like your survey scan (MS1 scan) is divided into a few events, and unfortunately MS-DIAL is not compatible with that kind of method. We will try to make it work, but I can not say any deadline...
We deeply apologize for the inconvinience.

Gio-Vi, 2025-05-22:
Hi Bujee415,
Thanks for taking the time to check the data. If you're able to get it working, that would be perfect for us - and hopefully beneficial for the community as well.
No need to apologize - this is how software improves :)
Please keep us posted if you have any updates.
Best regards,
Gio