systemsomicslab/MsdialWorkbench #561: Window Installer File not getting Installed

sidra-azhar, 2025-05-14:
Plzz let me know how I can open Window Installer file that I have downloaded from MSP spectral libraries ? Bz as I try to open... notification comes that  "File could not be opened.Asked application vendour that window installation patch is safe"

Bujee415, 2025-05-15:
Hello,
Can you tell me what you are trying to do specifically? 
Are you trying to use the MSP library on MS-DIAL? In that case, you can just add the MSP library from the "Identification" tab by clicking the "+" button. 
Thank you.

sidra-azhar, 2025-05-15:
First of all I thank you for you for your attention.
Actually I have sorted this issue, but now I have run my mzXML file in both
positive and negative mode separately but at the end no peak information
obtained. Bz of polarity mismatch.
Can you plz tell me how to resolve this problem?

Bujee415, 2025-05-15:
I thought that you analyzed it separately (pos and neg), but what do you mean because of polarity mismatch?
You need to analyze positive ion mode data in positive ion mode setting and negative in negative ion mode setting.
Is there anything that I could not understand?

sidra-azhar, 2025-05-15:
Yes, I have analyze positive ion mode in positive and negative ion mode in
negative setting but at the end notification comes stating "no peak
information due to polarity issue " and 2nd thing is that at a time we can
select one mode of analysis that's why I'm doing it separately.
Thanks

sidra-azhar, 2025-05-15:
I have sent you file in google drive. Plz check your mail.

Bujee415, 2025-05-21:
Hello,
It seems like your data has no MS (just MS/MS).
MS-DIAL uses MS for peak picking, so this is the reason.
Hope it helps!

closed 2025-05-21 by Bujee415