systemsomicslab/MsdialWorkbench #607: MS Method Type Not Appearing

samantha-swan-ocme, 2025-08-01:
Hello! I am analyzing QTOF SWATH data using MS-DIAL and realized that in 'Setting project parameters' my 'Measurement parameters' tab does not have the 'MS method type' selection. I am using MSDIAL.v5.5.250627 which I believe is the most recent version. I assigned the 'Acquisition type' to SWATH in my 'Raw measurement files' tab for upload but in my measurement parameters I only have the option to assign Ionization type, Separation type, Collision type, Data type, Ion mode, and Target omics. I tried the wiff files and the mzML files to see if that would impact anything.

Any help would be appreciated. Thank you!

[image]

samantha-swan-ocme, 2025-08-01:
Just to confirm version

[image]

biswapriyamisra, 2025-08-15:
Hey Samantha,
Thanks for sharing the issue.

Right after choosing your own data directory, in the "Raw Measurement Files" option you have to select SWATH to be able to specify and see AIF/SWATH settings. With default being "DDA" you do not see the Options, I believe.

Please retry.
Hope this helps.

Thanks again,
Biswa

[image]

closed 2026-01-30 by Bujee415