systemsomicslab/MsdialWorkbench #636: Unhandled exception in MSDial v5.5.250820

Lizeng-9, 2025-09-29:
When using the same dataset with MS-Dial version 5.5.250820, the program closed directly (see attached).
Has someone experienced a similar problem?
Or can someone interpret the error message more precisely?
Really thanks for your support.

[image]

wkumler, 2025-11-26:
Maybe related to #675?

Bujee415, 2026-01-30:
Hello!
Sorry for the late reply!

It seems like there are some issues with MS deconvolution.
Can you try it with our newest release and let us know if it works?
Also, if it doesn't work, please send us some demo data so that we can reproduce this issue and fix it?

Thank you,
Bujee